bagel99/llvm-my66000 #2: Storing the same constant multiple times

tkoenig1, 2022-02-20:
I had a few more looks at fatigue2.s from Issue#1, and found another thing that I
thought worth mentioning.

In the function read_input_m_ , there is a code section which reads

```
        std     #-4317352126650676160,[r18]
        mov     r18,r28
        .loc    1 267 1                         ; fatigue2.f90:267:1
        std     #-4317352126650676160,[r28]
        std     r26,[sp,216]
        .loc    1 268 1                         ; fatigue2.f90:268:1
        std     #-4317352126650676160,[r26]
        ldd     r2,[sp,272]
        .loc    1 269 1                         ; fatigue2.f90:269:1
        std     #-4317352126650676160,[r2]
        ldd     r25,[sp,280]
        ldd     r2,[sp,264]
        .loc    1 270 1                         ; fatigue2.f90:270:1
        std     #-4317352126650676160,[r2]
        std     r27,[sp,208]
        .loc    1 271 1                         ; fatigue2.f90:271:1
        std     #-4317352126650676160,[r27]
        std     r20,[sp,200]
        .loc    1 272 1                         ; fatigue2.f90:272:1
        std     #-4317352126650676160,[r20]
        ldd     r20,[sp,320]
        std     r19,[sp,192]
        .loc    1 273 1                         ; fatigue2.f90:273:1
        std     #-4317352126650676160,[r19]
        std     r16,[sp,184]
        .loc    1 274 1                         ; fatigue2.f90:274:1
        std     #-4317352126650676160,[r16]
```
(I have no reason why flang likes to store the same constant there over and over again, I just used it to generate test cases because it is the only Fortran compiler I have access which generates LLVM :-)

It would make sense to load the constant -4317352126650676160 into a register and just store it, to the different places it wants to be stored to, without putting it into the instruction stream multiple times.

The original Fortran code, by the way, is initializing quite a few variables with the same value, -1.D_20, it reads like

```
      plastic_strain_threshold = -1.0e20_LONGreal
      spin_frequency = -999
      Poissons_ratio = -1.0e20_LONGreal
      Youngs_modulus = -1.0e20_LONGreal
      shear_modulus = -1.0e20_LONGreal
      wire_test = .false.
      coil_test = .false.
      wire_radius = -1.0e20_LONGreal
      coil_radius = -1.0e20_LONGreal
      coil_pitch = -1.0e20_LONGreal
      radius_of_curvature = -1.0e20_LONGreal

```

tkoenig1, 2022-02-20:
A shorter test case:
```
void foo (double *a, double *b, double *c)
{
  *a = 42.;
  *b = 42.;
  *c = 42.;
}
```
gets translated to
```
foo:                                    ; @foo
        .cfi_startproc
; %bb.0:                                ; %entry
        std     #4631107791820423168,[r1]
        std     #4631107791820423168,[r2]
        std     #4631107791820423168,[r3]
        ret
```

bagel99, 2022-02-24:
Fixed in commit aca248511f643f5f50497e1545d8bb9e6e5d4907

closed 2022-02-24 by bagel99