bagel99/llvm-my66000 #3: Vectorizing "any"

tkoenig1, 2022-07-24:
The code
```
_Bool any( _Bool a[], long n)
{    
     for( long i = 0; i < n; i++ )
             if( a[i] )
                  return 1;
     return 0;
}
```
yields, with
```
$ clang -S -emit-llvm -O3 any.c
$ llc  -O3 --march=my66000 --frame-pointer=none --enable-vvm any.ll
```
the following assembly:
```
any:                                    ; @any
        .cfi_startproc
; %bb.0:                                ; %entry
        cmp     r3,r2,#1
        blt     r3,.LBB0_1
; %bb.2:                                ; %for.body.preheader
        ldub    r4,[r1]
        mov     r3,#1
        beq0    r4,.LBB0_3
; %bb.7:                                ; %cleanup
        and     r1,r3,#1
        ret
.LBB0_1:
        mov     r3,#0
        and     r1,r3,#1
        ret
.LBB0_3:                                ; %for.cond.preheader
        mov     r4,#1
.LBB0_4:                                ; %for.cond
                                        ; =>This Inner Loop Header: Depth=1
        mov     r3,r4
        cmp     r4,r2,r3
        beq     r4,.LBB0_6
; %bb.5:                                ; %for.body
                                        ;   in Loop: Header=BB0_4 Depth=1
        ldub    r5,[r1,r3,0]
        add     r4,r3,#1
        beq0    r5,.LBB0_4
.LBB0_6:                                ; %cleanup.loopexit
        cmp     r1,r3,r2
        srl     r3,r1,<1:4>
        and     r1,r3,#1
        ret
.Lfunc_end0:
        .size   any, .Lfunc_end0-any
        .cfi_endproc
                                        ; -- End function
        .ident  "clang version 12.0.0"
        .section        ".note.GNU-stack","",@progbits
```
without any `vec` or `loop` instructions.

bagel99, 2022-11-24:
VVM loops currently cannot have any control flow, except predicates.

closed 2022-11-30 by bagel99

tkoenig1, 2022-12-27:
Not sure that this is not supported in VVM.

Mitch stated in [https://groups.google.com/g/comp.arch/c/ROo8YXJPVa0/m/wDbyGPFjCAAJ](url) that, eventually, he would eventually want
```
any:
          MOV     R3,#0
          MOV     R7,R1               // switch registers so R1 carries #1 return code
          MOV     R1,#1
          CMP      R4,R3,R2
          BGE       R4,exit
          VEC       R5,{}
          LDUB     R6,[R7+R3]
          PNE       R6,T
          RET
          LOOP    LT,R3,#1,R2
exit:
          MOV     R1,#0
          RET
```