bagel99/llvm-my66000 #5: Internal compiler error "Cannot print this instruction"

tkoenig1, 2022-10-28:
The following bc code, translated from the Livermore ODEPACK, yields an internal compiler error, with "Cannot print this instructions". bc code and output are attached.

[bug.zip](https://github.com/bagel99/llvm-my66000/files/9889741/bug.zip)

tkoenig1, 2022-10-28:
Command line was

```
nohup llc --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 opkda1_opt.bc
```

bagel99, 2022-11-25:
The cause appears to be the use of vectors in the IR, e.g.:
  %.splatinsert = insertelement <4 x i32> undef, i32 %33, i32 0, !dbg !138
Possibly a problem with header files.

tkoenig1, 2022-11-26:
No header files, the original source code is Fortran :-)

Anyway, I ran across the same error in GSL.  I have attached a reduced test case, wich is just plain C, to reproduce the error.

[print_this_instruction.zip](https://github.com/bagel99/llvm-my66000/files/10096301/print_this_instruction.zip)

bagel99, 2022-11-26:
Are you using flang?  If so, what target architecture?

tkoenig1, 2022-11-27:
> Are you using flang? If so, what target architecture?

I run flang on x86_64:

```
$ flang --version
clang version 7.0.1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

I'm not quite sure why this matters, though.  Isn't LLVM supposed to be target-independent?

bagel99, 2022-11-27:
The front ends for LLVM are NOT target-independent.  (It took me a while to realize that.)  Nor is "opt" for that matter.  So when using clang for My66000, you must use the clang built from my tree because it was modified to add my66000 as a target.  And you must use the target flag:
  clang --target=my66000 ...
As for flang, I have not modified that yet.  A quick glance at the release 15 sources shows, I think, that the machine dependent code is just about how to pass and return complex values (pairs): 1. as a vectors, 2. as an arrary, 3. as a struct.  Playing with different targets may yield a version that my66000 is happy with.

tkoenig1, 2022-11-29:
If you look at flang, the best way is to be compatible with C.  This usually has
```
typedef struct { real r, i; } complex;
```
(following f2c).

bagel99, 2022-11-30:
Looking at the flang source flang/lib/Optimizer/CodeGen/Target.cpp the only 64-bit targets are 
  case llvm::Triple::ArchType::x86_64:
  case llvm::Triple::ArchType::aarch64:
  case llvm::Triple::ArchType::ppc64le:
  case llvm::Triple::ArchType::sparcv9:
I would try each to see what happens when the backend targets my66000.

bagel99, 2022-11-30:
Should be fixed in commit a049e4a8ace12314d0681801797069ee02fd0a5d

bagel99, 2022-12-01:
Fixed in commit a049e4a8ace12314d0681801797069ee02fd0a5d.

closed 2022-12-01 by bagel99