bagel99/llvm-my66000 #6: LLVM error: Cannot select

tkoenig1, 2022-11-18:
Compiling the attached file from GSL 2.7 yields the error

```
VVMLoopPass: gsl_odeiv_evolve_free
LLVM ERROR: Cannot select: t26: f64 = select t81, t25, t20
  t81: i64 = and t22, Constant:i64<1>
    t22: i64,ch = CopyFromReg t0, Register:i64 %12
      t21: i64 = Register %12
    t8: i64 = Constant<1>
  t25: f64,ch = CopyFromReg t0, Register:f64 %31
    t24: f64 = Register %31
  t20: f64 = fadd t19, t12
    t19: f64,ch = CopyFromReg t0, Register:f64 %0
      t18: f64 = Register %0
    t12: f64,ch = load<(dereferenceable load 8 from %ir.h0, !tbaa !13)> t10, FrameIndex:i64<0>, undef:i64
      t11: i64 = FrameIndex<0>
      t6: i64 = undef
In function: gsl_odeiv_evolve_apply
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 evolve_opt.bc
1.      Running pass 'Function Pass Manager' on module 'evolve_opt.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@gsl_odeiv_evolve_apply'
 #0 0x0000000014aa5b60 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x0000000014aa5c98 PrintStackTraceSignalHandler(void*) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:631:23
 #2 0x0000000014aa2ec4 llvm::sys::RunSignalHandlers() /home/tkoenig/llvm-my66000/llvm/lib/Support/Signals.cpp:70:22
 #3 0x0000000014aa5148 SignalHandler(int) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:397:31
```
[gsl-evolve.zip](https://github.com/bagel99/llvm-my66000/files/10045477/gsl-evolve.zip)

The command used for compilation is a simplistic script:
```
a=${1%%.c}
b=${a}_opt
clang -I .. -c -O3 -emit-llvm -fno-unroll-loops $a.c
opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm $a.bc  > $b.bc
llc --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 $b.bc
```

tkoenig1, 2022-11-19:
[evolve-preprocessed.zip](https://github.com/bagel99/llvm-my66000/files/10047553/evolve-preprocessed.zip)

Probably better with the preprocessed source.

bagel99, 2022-11-26:
Fixed in ec44be29269979cc5e9acb90ec6967bff8b57033.

closed 2022-11-26 by bagel99