bagel99/llvm-my66000 #7: Internal compiler error: LowerCall emitted a return value for a tail call!"' failed.

tkoenig1, 2022-11-18:
Another issue with GSL.   Source file and bc attached.

[gsl-interp2d.zip](https://github.com/bagel99/llvm-my66000/files/10045538/gsl-interp2d.zip)

Error message:

```
VMLoopPass: gsl_interp2d_alloc
VVMLoopPass: gsl_interp2d_free
VVMLoopPass: gsl_interp2d_init
VVMLoopPass: gsl_interp2d_eval
VVMLoopPass: gsl_interp2d_eval_e
VVMLoopPass: gsl_interp2d_eval_extrap
llc: /home/tkoenig/llvm-my66000/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:9453: std::pair<llvm::SDValue, llvm::SDValue> llvm::TargetLowering::LowerCallTo(CallLoweringInfo&) const: Assertion `(!CLI.IsTailCall || InVals.empty()) && "LowerCall emitted a return value for a tail call!"' failed.
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 interp2d_opt.bc
1.      Running pass 'Function Pass Manager' on module 'interp2d_opt.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@gsl_interp2d_eval_e_extrap'
 #0 0x0000000014aa5b60 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x0000000014aa5c98 PrintStackTraceSignalHandler(void*) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:631:23
 #2 0x0000000014aa2ec4 llvm::sys::RunSignalHandlers() /home/tkoenig/llvm-my66000/llvm/lib/Support/Signals.cpp:70:22
 #3 0x0000000014aa5148 SignalHandler(int) /home/tkoenig/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:397:31
 #4 0x00007fffb43c04d8  0x4d8 abort

```

tkoenig1, 2022-11-18:
Code of the offending function is
```
#define DISCARD_STATUS(s) if ((s) != GSL_SUCCESS) { GSL_ERROR_VAL("interpolation error", (s),  GSL_NAN); }
```
...
```
double
gsl_interp2d_eval_extrap (const gsl_interp2d * interp,
                          const double xarr[],
                          const double yarr[],
                          const double zarr[],
                          const double x,
                          const double y,
                          gsl_interp_accel * xa,
                          gsl_interp_accel * ya)
{
  double z;
  int status =
    interp2d_eval_extrap(interp->type->eval, interp,
                         xarr, yarr, zarr, x, y, xa, ya, &z);
  DISCARD_STATUS(status)
  return z;
}
```

tkoenig1, 2022-11-18:
and
```
#define GSL_ERROR_VAL(reason, gsl_errno, value) \
       do { \
       gsl_error (reason, __FILE__, __LINE__, gsl_errno) ; \
       return value ; \
       } while (0)
```

tkoenig1, 2022-11-19:
Somewhat reduced test case:
```
typedef long unsigned int size_t;
extern int *__errno_location (void) __attribute__ ((__nothrow__ , __leaf__)) __attribute__ ((__const__));
enum {
  GSL_SUCCESS = 0,
  GSL_FAILURE = -1,
  GSL_CONTINUE = -2,
  GSL_EDOM = 1,
  GSL_ERANGE = 2,
  GSL_EFAULT = 3,
  GSL_EINVAL = 4,
  GSL_EFAILED = 5,
  GSL_EFACTOR = 6,
  GSL_ESANITY = 7,
  GSL_ENOMEM = 8,
  GSL_EBADFUNC = 9,
  GSL_ERUNAWAY = 10,
  GSL_EMAXITER = 11,
  GSL_EZERODIV = 12,
  GSL_EBADTOL = 13,
  GSL_ETOL = 14,
  GSL_EUNDRFLW = 15,
  GSL_EOVRFLW = 16,
  GSL_ELOSS = 17,
  GSL_EROUND = 18,
  GSL_EBADLEN = 19,
  GSL_ENOTSQR = 20,
  GSL_ESING = 21,
  GSL_EDIVERGE = 22,
  GSL_EUNSUP = 23,
  GSL_EUNIMPL = 24,
  GSL_ECACHE = 25,
  GSL_ETABLE = 26,
  GSL_ENOPROG = 27,
  GSL_ENOPROGJ = 28,
  GSL_ETOLF = 29,
  GSL_ETOLX = 30,
  GSL_ETOLG = 31,
  GSL_EOF = 32
} ;
void gsl_error (const char * reason, const char * file, int line,
                int gsl_errno);

enum
  {
    FP_NAN =
      0,
    FP_INFINITE =
      1,
    FP_ZERO =
      2,
    FP_SUBNORMAL =
      3,
    FP_NORMAL =
      4
  };

typedef struct {
  size_t cache;
  size_t miss_count;
  size_t hit_count;
}
gsl_interp_accel;

typedef struct {
  const char * name;
  unsigned int min_size;
  void * (*alloc) (size_t size);
  int (*init) (void *, const double xa[], const double ya[], size_t size);
[... 125 lines cut ...]
  return interp2d_eval(interp->type->eval, interp,
                       xarr, yarr, zarr, x, y, xa, ya, z);
}

int
gsl_interp2d_eval_e_extrap (const gsl_interp2d * interp,
                            const double xarr[], const double yarr[],
                            const double zarr[], const double x,
                            const double y, gsl_interp_accel * xa,
                            gsl_interp_accel * ya, double * z)
{
  return interp2d_eval_extrap(interp->type->eval, interp,
                              xarr, yarr, zarr, x, y, xa, ya, z);
}
int
gsl_interp2d_eval_extrap_e (const gsl_interp2d * interp,
                            const double xarr[], const double yarr[],
                            const double zarr[], const double x,
                            const double y, gsl_interp_accel * xa,
                            gsl_interp_accel * ya, double * z)
{
  return interp2d_eval_extrap(interp->type->eval, interp,
                              xarr, yarr, zarr, x, y, xa, ya, z);
}
```

tkoenig1, 2022-11-29:
This issue persists when compiled for my66000.

bagel99, 2022-11-30:
Should be fixed in commit a049e4a8ace12314d0681801797069ee02fd0a5d

bagel99, 2022-12-01:
Fixed in commit a049e4a8ace12314d0681801797069ee02fd0a5d.

closed 2022-12-01 by bagel99