bagel99/llvm-my66000 #13: A few gnuplot segfaults

tkoenig1, 2022-12-02:
[gnuplot_bugs.zip](https://github.com/bagel99/llvm-my66000/files/10144930/gnuplot_bugs.zip)

Hi, here are a few gnuplot bugs which cause segementation faults.  It looks like they might be the same error, bu I'm not sure.

Error messages are
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 axis_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'axis_dbg.bc'.
2.      Running pass 'My66000 Assembly Printer' on function '@clone_linked_axes'
#0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
#1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
#2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
#3 0x00007f58a43098c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
#4 0x00007ffc70e34600
```
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 datafile_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'datafile_dbg.bc'.
2.      Running pass 'My66000 Assembly Printer' on function '@df_open'
#0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
#1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
#2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
#3 0x00007f65d7ab98c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
#4 0x00007fffd13f2b00
```
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 external_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'external_dbg.bc'.
2.      Running pass 'My66000 Assembly Printer' on function '@external_at'
#0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
#1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
#2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
#3 0x00007faa26db18c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
#4 0x00007ffee546d1b0
```

bagel99, 2022-12-20:
axis_dbg.bc - unable to duplicate
datafile_dbg.bc - unable to duplicate
external_dbg.bc - unable to duplicate
Try running the latest commit.

tkoenig1, 2022-12-20:
The minimum options I need for axis_dbg.bc is
```
tkoenig@gcc188:~/GNUPLOT> llc --enable-predication2 --enable-predication -march=my66000 axis_dbg.bc 
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --enable-predication2 --enable-predication -march=my66000 axis_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'axis_dbg.bc'.
2.      Running pass 'My66000 Assembly Printer' on function '@clone_linked_axes'
#0 0x00000000026da51c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da51c)
#1 0x00000000026d83f4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d83f4)
#2 0x00000000026d8553 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8553)
#3 0x00007f9a234208c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
#4 0x00007ffc6dcdd880 
Speicherzugriffsfehler (Speicherabzug geschrieben)
```
This is with
```
clang version 12.0.0 (https://github.com/bagel99/llvm-my66000.git cd67543d12f7168fb0e502c7da98dc1db7777b9e)
```
which is the latest commit.

I have been trying to debug llc, but I didn't get gdb to find the sources.

bagel99, 2022-12-21:
OK.  I was not using --enable-predication2.  That part of the my66000 backend is very buggy and incomplete.  Try all your tests without that flag.
And remove the -g flag from clang.  It just produces huge .ll and .bc files filled with debug cruft we don't yet deal with.  Smaller .ll and .bc files will help in the bug hunting.  Thanks.

tkoenig1, 2022-12-22:
With --enable-predication2 gone, I am getting the same results as you are seeing, a single failure in set.i.

I have to see about getting one of these automated reduction tools installed...

tkoenig1, 2022-12-24:
The failure in set.i is now also gone in clang version 12.0.0 (https://github.com/bagel99/llvm-my66000.git 242c042a9d39f8e2aa5ff43d964fc531474dc185)

I will try to play around with cvise to get a reduced test case with --enable-predication2 (just to learn the tool :-)

bagel99, 2023-01-04:
I think these are all fixed by commit acfbd5e43e21737020a6a9a320d7a764426372ed.

bagel99, 2023-01-05:
Fixed by commit commit acfbd5e43e21737020a6a9a320d7a764426372ed.

closed 2023-01-05 by bagel99