bagel99/llvm-my66000 #14: A few more gnuplot internal compiler errors

tkoenig1, 2022-12-03:
These have different error messages to Issue #13, may or may not be the same underlying cause.
[gpbug2.zip](https://github.com/bagel99/llvm-my66000/files/10146137/gpbug2.zip)
Error messages:
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 interpol_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'interpol_dbg.bc'.
2.      Running pass 'My66000 predicate transform pass' on function '@make_bins'
 #0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
 #1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
 #2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
 #3 0x00007fc1518708c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
 #4 0x0000000001bd3b86 llvm::MachineBasicBlock::removePredecessor(llvm::MachineBasicBlock*) (/home/tkoenig/bin/llc+0x1bd3b86)
 #5 0x0000000001bd536c llvm::MachineBasicBlock::removeSuccessor(__gnu_cxx::__normal_iterator<llvm::MachineBasicBlock**, std::vector<llvm::MachineBasicBlock*, std::allocator<llvm::MachineBasicBlock*> > >, bool) (/home/tkoenig/bin/llc+0x1\
bd536c)
 #6 0x00000000006a853c (anonymous namespace)::My66000PredBlock::ConvertD2(llvm::MachineBasicBlock*, llvm::MachineBasicBlock*, llvm::MachineBasicBlock*, llvm::MachineBasicBlock*, llvm::MachineBasicBlock*) (/home/tkoenig/bin/llc+0x6a853c)
 #7 0x000000000161b136 (anonymous namespace)::My66000PredBlock::runOnMachineFunction(llvm::MachineFunction&) (/home/tkoenig/bin/llc+0x161b136)
 #8 0x0000000001c2de68 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1c2de68)
 #9 0x0000000001ff977f llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1ff977f)
#10 0x0000000001ff9e61 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff9e61)
#11 0x0000000001ff8b47 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff8b47)
#12 0x000000000079742b compileModule(char**, llvm::LLVMContext&) (/home/tkoenig/bin/llc+0x79742b)
#13 0x00000000006d7872 main (/home/tkoenig/bin/llc+0x6d7872)
#14 0x00007fc1502d129d __libc_start_main (/lib64/libc.so.6+0x3529d)
#15 0x000000000078f42a _start /home/abuild/rpmbuild/BUILD/glibc-2.31/csu/../sysdeps/x86_64/start.S:122:0
```
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 set_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'set_dbg.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@set_tic_prop'
 #0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
 #1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
 #2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
 #3 0x00007f40211bb8c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
 #4 0x0000000001622bbf (anonymous namespace)::My66000DAGToDAGISel::Select(llvm::SDNode*) (/home/tkoenig/bin/llc+0x1622bbf)
 #5 0x0000000002533c64 llvm::SelectionDAGISel::DoInstructionSelection() (/home/tkoenig/bin/llc+0x2533c64)
 #6 0x000000000253b4e6 llvm::SelectionDAGISel::CodeGenAndEmitDAG() (/home/tkoenig/bin/llc+0x253b4e6)
 #7 0x000000000253e96c llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (/home/tkoenig/bin/llc+0x253e96c)
 #8 0x0000000002540ba8 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (.part.845) (/home/tkoenig/bin/llc+0x2540ba8)
 #9 0x0000000001c2de68 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1c2de68)
#10 0x0000000001ff977f llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1ff977f)
#11 0x0000000001ff9e61 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff9e61)
#12 0x0000000001ff8b47 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff8b47)
#13 0x000000000079742b compileModule(char**, llvm::LLVMContext&) (/home/tkoenig/bin/llc+0x79742b)
#14 0x00000000006d7872 main (/home/tkoenig/bin/llc+0x6d7872)
#15 0x00007f401fc1c29d __libc_start_main (/lib64/libc.so.6+0x3529d)
#16 0x000000000078f42a _start /home/abuild/rpmbuild/BUILD/glibc-2.31/csu/../sysdeps/x86_64/start.S:122:0
```
```
PLEASE submit a bug report to https://bugs.llvm.org/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 graphics_dbg.bc
1.      Running pass 'Function Pass Manager' on module 'graphics_dbg.bc'.
2.      Running pass 'Branch Probability Basic Block Placement' on function '@place_objects'
 #0 0x00000000026da30c llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/tkoenig/bin/llc+0x26da30c)
 #1 0x00000000026d81e4 llvm::sys::RunSignalHandlers() (/home/tkoenig/bin/llc+0x26d81e4)
 #2 0x00000000026d8343 SignalHandler(int) (/home/tkoenig/bin/llc+0x26d8343)
 #3 0x00007f0bb8ca68c0 __restore_rt (/lib64/libpthread.so.0+0x168c0)
 #4 0x0000000001bea5c2 (anonymous namespace)::MachineBlockPlacement::markBlockSuccessors((anonymous namespace)::BlockChain const&, llvm::MachineBasicBlock const*, llvm::MachineBasicBlock const*, llvm::SmallSetVector<llvm::MachineBasicBl\
ock const*, 16u> const*) (/home/tkoenig/bin/llc+0x1bea5c2)
 #5 0x0000000001bee640 (anonymous namespace)::MachineBlockPlacement::buildChain(llvm::MachineBasicBlock const*, (anonymous namespace)::BlockChain&, llvm::SmallSetVector<llvm::MachineBasicBlock const*, 16u>*) (/home/tkoenig/bin/llc+0x1be\
e640)
 #6 0x0000000001bf310f (anonymous namespace)::MachineBlockPlacement::buildCFGChains() (/home/tkoenig/bin/llc+0x1bf310f)
 #7 0x0000000001bf3838 (anonymous namespace)::MachineBlockPlacement::runOnMachineFunction(llvm::MachineFunction&) (.part.384) (/home/tkoenig/bin/llc+0x1bf3838)
 #8 0x0000000001c2de68 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1c2de68)
 #9 0x0000000001ff977f llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/tkoenig/bin/llc+0x1ff977f)
#10 0x0000000001ff9e61 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff9e61)
#11 0x0000000001ff8b47 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/tkoenig/bin/llc+0x1ff8b47)
#12 0x000000000079742b compileModule(char**, llvm::LLVMContext&) (/home/tkoenig/bin/llc+0x79742b)
#13 0x00000000006d7872 main (/home/tkoenig/bin/llc+0x6d7872)
#14 0x00007f0bb770729d __libc_start_main (/lib64/libc.so.6+0x3529d)
#15 0x000000000078f42a _start /home/abuild/rpmbuild/BUILD/glibc-2.31/csu/../sysdeps/x86_64/start.S:122:0
```

bagel99, 2022-12-20:
interpol_dbg.bc - unable to duplicate
graphics_dbg.bc - unable to duplicate
set_dbg.bc - asserts while processing routine set_tic_prop
This routine is over 17000 lines into the preprocessed C file set.i.  Can you cut down the size of the input file while leaving set_tic_prop?

bagel99, 2022-12-22:
interpol_dbg.bc - workaround: do not use flag --enable-predication2
graphics_dbg.bc - workaround: do not use flag --enable-predication2
set_dbg.bc - fixed in commit 242c042a9d39f8e2aa5ff43d964fc531474dc185

tkoenig1, 2022-12-25:
Here's a reduction of interpol.i:

```
double min;
struct axis {
  double max
} * make_bins_nbins_yaxis;
axis_array[];
make_bins_nbins_i, make_bins_nbins_ymax;
*make_bins_nbins_bin;
make_bins_nbins() {
  struct axis *xaxis = axis_array;
  for (;;)
    if (xaxis->max ? 0 >= 0 : 0 >= xaxis && min)
      make_bins_nbins_ymax = make_bins_nbins_bin[make_bins_nbins_i];
    else
      make_bins_nbins_yaxis->max = make_bins_nbins_ymax;
}
```
This was reduced using cvise with the helper script
```
clang -O2 -g -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops interpol.i || exit 1
opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm interpol.bc > interpol_opt.bc
llc -O2 --disable-lsr --enable-predication --enable-predication2 -march=my66000 interpol_opt.bc 2>&1  | grep runOnFunction
```

Expect shorter bug reports from me in the future :-)

bagel99, 2022-12-27:
Perhaps a little too reduced :-)  This doesn't compile for me, maybe a cut/paste error?

tkoenig1, 2022-12-27:
Maybe this will work better.

It also contains a different reduction with --enable-predication --enable-predication2, this time from Perl.

[ice.zip](https://github.com/bagel99/llvm-my66000/files/10311015/ice.zip)

Helper script for toke.i was

```
clang -O2 -g -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops toke.i || exit 1
opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm toke.bc > toke_opt.bc
llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 toke_opt.bc 2>&1 | grep removePredecessor
```

bagel99, 2022-12-28:
Not better -- toke.i doesn't compile.

tkoenig1, 2022-12-29:
[toke.zip](https://github.com/bagel99/llvm-my66000/files/10318397/toke.zip)

> Not better -- toke.i doesn't compile.

Weird. What are the error messages?

Anyway, here are two reductions of the file (one by adding types by hand, one by running cvise with -Werror). Here is also something I reduced a little bit further by hand:
```
int a, b, c, d;
_Bool foo();
int bar() {
  int e = 0;
  while (foo()) {
    e += a || b && c ? d : 1;
  }
  return e;
}
```
I hope that at least the last one compiles for you.

bagel99, 2022-12-29:
That one does compile, but does not trigger any bugs.  I have reduced interpol.i routine make_bins to this:
struct axis
{   double min;
    double max;
};

int type;

double foo(struct axis *xaxis, double bincent, double ybin, double ymax)
{
     if ((((xaxis->min)<(xaxis->max))
         ? (((bincent)>=(xaxis->min)) && ((bincent)<=(xaxis->max))) 
         : (((bincent)>=(xaxis->max)) && ((bincent)<=(xaxis->min))))) {
       if (ymax < ybin)
         ymax = ybin;
    } else {
      type = 1;
    }
    return ymax;
}

This triggers a bug in the predication2 code.  There probably are not enough checks for a sane layout of basic blocks
to allow a predication conversion.

tkoenig1, 2022-12-30:
So, either non-deterministic basic block ordering, or dependent on special options.

For future reference: What happens if you use the exact same shell commands I posted above as a test script? Reducing makes little sense if you cannot reproduce it :-|

bagel99, 2023-01-04:
I think these are all fixed by commit acfbd5e43e21737020a6a9a320d7a764426372ed.

tkoenig1, 2023-01-04:
Both Perl and Gnuplot are compiled to assembly without an ICE. Great!

bagel99, 2023-01-05:
Thanks for testing.
Fixed by commit acfbd5e43e21737020a6a9a320d7a764426372ed.

closed 2023-01-05 by bagel99