bagel99/llvm-my66000 #17: CBLAS sswap not vectorized

tkoenig1, 2023-01-07:
This is from GSL, which in turn uses CBLAS.

The function
```
void
cblas_sswap (const int N, float *X, const int incX, float *Y, const int incY)
{

  int i;
  int ix = ((incX) > 0 ? 0 : ((N) - 1) * (-(incX)));
  int iy = ((incY) > 0 ? 0 : ((N) - 1) * (-(incY)));
  for (i = 0; i < N; i++)
    {
      const float tmp = X[ix];
      X[ix] = Y[iy];
      Y[iy] = tmp;
      ix += incX;
      iy += incY;
    }
}
```
is currently not vectorized, the assembly for loops is
```
cblas_sswap:                            ; @cblas_sswap
; %bb.0:
        ble0    r1,.LBB0_5
; %bb.1:
        add     r6,#1,-r1
        mul     r7,r6,r5
        cmp     r8,r5,#0
        sra     r8,r8,<1:2>
        mov     r9,#0
        mux     r7,r8,r9,r7
        mul     r6,r6,r3
        cmp     r8,r3,#0
        sra     r8,r8,<1:2>
        mux     r8,r8,r9,r6
        sra     r6,r7,<32:0>
        sra     r7,r8,<32:0>
        lduw    r8,[r2,r7<<2,0]
        lduw    r9,[r4,r6<<2,0]
        stw     r9,[r2,r7<<2,0]
        stw     r8,[r4,r6<<2,0]
        cmp     r8,r1,#1
        bne     r8,.LBB0_2
.LBB0_5:                                ; %.loopexit
        ret
.LBB0_2:                                ; %..lr.ph_crit_edge
        mov     r8,#2
.LBB0_3:                                ; %.lr.ph
                                        ; =>This Inner Loop Header: Depth=1
        add     r6,r6,r5
        add     r7,r7,r3
        lduw    r9,[r2,r7<<2,0]
        lduw    r10,[r4,r6<<2,0]
        stw     r10,[r2,r7<<2,0]
        stw     r9,[r4,r6<<2,0]
        sra     r9,r8,<32:0>
        cmp     r9,r9,r1
        bge     r9,.LBB0_5
; %bb.4:                                ; %.lr.ph..lr.ph_crit_edge
                                        ;   in Loop: Header=BB0_3 Depth=1
        add     r8,r8,#1
        br      .LBB0_3
```
Script used for compilation is
```
#! /bin/bash
a=${1%%.[ci]}
b=${a}_opt
clang -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops $1
opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm $a.bc  > $b.bc
llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 $b.bc
```
which is unrolled two times, but not vectorized.

tkoenig1, 2023-01-07:
Maybe related to / the same issue as #4 .

bagel99, 2023-01-19:
I can not duplicate.  I ran with your compilation script and got:
cblas_sswap:                            ; @cblas_sswap
; %bb.0:
	ble0	r1,.LBB0_3
; %bb.1:
	add	r7,#1,-r1
	mul	r8,r7,r5
	cmp	r6,r5,#0
	sra	r9,r6,<1:2>
	mov	r6,#0
	mux	r8,r9,r6,r8
	mul	r7,r7,r3
	cmp	r9,r3,#0
	sra	r9,r9,<1:2>
	mux	r9,r9,r6,r7
	sra	r7,r8,<32:0>
	sra	r8,r9,<32:0>
.LBB0_2:                                ; =>This Inner Loop Header: Depth=1
	vec	r9,{}
	lduw	r10,[r2,r8<<2,0]
	lduw	r11,[r4,r7<<2,0]
	stw	r11,[r2,r8<<2,0]
	stw	r10,[r4,r7<<2,0]
	add	r8,r8,r3
	add	r7,r7,r5
	add	r6,r6,#1
	sra	r6,r6,<32:0>
	loop	lt,r6,r1,#0
.LBB0_3:                                ; %.loopexit
	ret