bagel99/llvm-my66000 #19: __label__ not implemented

tkoenig1, 2023-01-22:
I am just running through the gcc testsuite, submitting one issue for each kind of failure.
```
 cat 20000518-1.c 
/* { dg-require-effective-target label_values } */

void callit1(void*);

extern __inline__ void test()
{
        __label__ l1;

        callit1(&&l1);

l1:;

}

void dotest()
{
        test();
}
```
yields
```
LLVM ERROR: Cannot select: 0x5597709390d8: i64 = BlockAddress<@test, %1> 0
In function: test
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 20000518-1_opt.bc
1.      Running pass 'Function Pass Manager' on module '20000518-1_opt.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@test'
 #0 0x000055976ce0f65f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/asdf/bin/llc+0x2ee365f)
 #1 0x000055976ce0d04e SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f9de081a420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #3 0x00007f9de02b700b raise /build/glibc-SzIz7B/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #4 0x00007f9de0296859 abort /build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c:81:7
 #5 0x000055976cd50347 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/asdf/bin/llc+0x2e24347)
 #6 0x000055976cbf35a0 llvm::SelectionDAGISel::CannotYetSelect(llvm::SDNode*) (/home/asdf/bin/llc+0x2cc75a0)
 #7 0x000055976cbf8e22 llvm::SelectionDAGISel::SelectCodeCommon(llvm::SDNode*, unsigned char const*, unsigned int) (/home/asdf/bin/llc+0x2ccce22)
 #8 0x000055976ba6d4d7 (anonymous namespace)::My66000DAGToDAGISel::Select(llvm::SDNode*) My66000ISelDAGToDAG.cpp:0:0
 #9 0x000055976cbf23b0 llvm::SelectionDAGISel::DoInstructionSelection() (/home/asdf/bin/llc+0x2cc63b0)
#10 0x000055976cbfd409 llvm::SelectionDAGISel::CodeGenAndEmitDAG() (/home/asdf/bin/llc+0x2cd1409)
#11 0x000055976cc00061 llvm::SelectionDAGISel::SelectAllBasicBlocks(llvm::Function const&) (/home/asdf/bin/llc+0x2cd4061)
#12 0x000055976cc02be6 llvm::SelectionDAGISel::runOnMachineFunction(llvm::MachineFunction&) (.part.0) SelectionDAGISel.cpp:0:0
#13 0x000055976c158cfc llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x222ccfc)
#14 0x000055976c5c6b90 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x269ab90)
#15 0x000055976c5c6d09 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/asdf/bin/llc+0x269ad09)
#16 0x000055976c5c8900 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/asdf/bin/llc+0x269c900)
#17 0x000055976a88baeb compileModule(char**, llvm::LLVMContext&) llc.cpp:0:0
#18 0x000055976a7a6782 main (/home/asdf/bin/llc+0x87a782)
```
I guess this is just a case of labels as variables not being imlemented.

bagel99, 2023-01-22:
Can duplicate.  (I hate languages that can have labels as variables.)  Will fix soon.

bagel99, 2023-01-23:
Test again with commit 6385ec062ccdd3350dcf89f4f2a3653449fb075c

tkoenig1, 2023-01-24:
> Test again with commit [6385ec0](https://github.com/bagel99/llvm-my66000/commit/6385ec062ccdd3350dcf89f4f2a3653449fb075c)

Looks good; also the generated code looks reasonable:
```
test:                                   ; @test
; %bb.0:                                ; %entry
        enter   r0,r0,0,0
        la      r1,[ip,.Ltmp0]
        call    callit1
.Ltmp0:                                 ; Block address taken
; %bb.1:                                ; %l1
        exit    r0,r0,0,0
.Lfunc_end0:
        .size   test, .Lfunc_end0-test
                                        ; -- End function
        .globl  dotest                          ; -- Begin function dotest
        .type   dotest,@function
dotest:                                 ; @dotest
; %bb.0:                                ; %entry
        br      test
```
Closing.

closed 2023-01-24 by tkoenig1