bagel99/llvm-my66000 #22: Apparent endless loop in compilation of pr44942.c

tkoenig1, 2023-01-22:
The compilation of a gcc test case took > 19 minutes on my home box before I killed it.
Here is the source:
```
/* PR target/44942 */

#include <stdarg.h>

void
test1 (int a, int b, int c, int d, int e, int f, int g, long double h, ...)
{
  int i;
  va_list ap;

  va_start (ap, h);
  i = va_arg (ap, int);
  if (i != 1234)
    __builtin_abort ();
  va_end (ap);
}

void
test2 (int a, int b, int c, int d, int e, int f, int g, long double h, int i,
       long double j, int k, long double l, int m, long double n, ...)
{
  int o;
  va_list ap;

  va_start (ap, n);
  o = va_arg (ap, int);
  if (o != 1234)
    __builtin_abort ();
  va_end (ap);
}

void
test3 (double a, double b, double c, double d, double e, double f,
       double g, long double h, ...)
{
  double i;
  va_list ap;

  va_start (ap, h);
  i = va_arg (ap, double);
  if (i != 1234.0)
    __builtin_abort ();
  va_end (ap);
}

void
test4 (double a, double b, double c, double d, double e, double f, double g,
       long double h, double i, long double j, double k, long double l,
       double m, long double n, ...)
{
  double o;
  va_list ap;

  va_start (ap, n);
  o = va_arg (ap, double);
  if (o != 1234.0)
    __builtin_abort ();
  va_end (ap);
}

int
main ()
{
  test1 (0, 0, 0, 0, 0, 0, 0, 0.0L, 1234);
  test2 (0, 0, 0, 0, 0, 0, 0, 0.0L, 0, 0.0L, 0, 0.0L, 0, 0.0L, 1234);
  test3 (0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0L, 1234.0);
  test4 (0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0L, 0.0, 0.0L,
         0.0, 0.0L, 0.0, 0.0L, 1234.0);
  return 0;
}
```
I have attached the *.bc file as well.

[pr44942.zip](https://github.com/bagel99/llvm-my66000/files/10474245/pr44942.zip)

tkoenig1, 2023-01-22:
Seems to be a more general problem with varargs, I had to kill several other test cases as well.

tkoenig1, 2023-01-22:
Shorter test case:
```
#include <stdarg.h>

typedef double L;
void f (L p0, L p1, L p2, L p3, L p4, L p5, L p6, L p7, L p8, ...)
{
  va_list select;

  va_start (select, p8);

  if (va_arg (select, int) != 10)
    abort ();
  if (va_arg (select, int) != 11)
    abort ();
  if (va_arg (select, int) != 12)
    abort ();

  va_end (select);
}

int main ()
{
  f (1., 2., 3., 4., 5., 6., 7., 8., 9., 10, 11, 12);
  exit (0);
}
```

tkoenig1, 2023-01-22:
With the Debug version, the error message changes to
```
Too many var args parameters.
UNREACHABLE executed at /home/asdf/llvm-my66000/llvm/lib/Target/My66000/My66000ISelLowering.cpp:935!
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
```

bagel99, 2023-01-31:
Try again with commit 4c3bbcf161cfd1940c03640c831d1e2b24ba783d

tkoenig1, 2023-02-01:
The internal compiler error is gone, but the code looks wrong.  I don't have a simulator to test it on, but the simplified test case 
```
/* PR target/44942 */

#include <stdarg.h>

void
test1 (int a, int b, int c, int d, int e, int f, int g, long double h, ...)
{
  int i;
  va_list ap;

  va_start (ap, h);
  i = va_arg (ap, int);
  if (i != 1234)
    __builtin_abort ();
  va_end (ap);
}

int
main ()
{
  test1 (0, 0, 0, 0, 0, 0, 0, 0.0L, 1234);
  return 0;
}
```
using the compile script
```
#! /bin/bash
a=${1%%.[ci]}
b=${a}_opt
clang -fomit-frame-pointer -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops $1
opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm $a.bc  > $b.bc
llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 $b.bc
```
results in the assembly code
```
test1:                                  ; @test1
; %bb.0:                                ; %entry
        add     sp,sp,#-16
        add     r1,sp,#0
        add     r1,r1,#8
        std     r1,[sp]
        and     r1,r1,#4294967292
        cmp     r1,r1,#1234
        bne     r1,.LBB0_1
; %bb.2:                                ; %if.end
        add     sp,sp,#16
        ret
.LBB0_1:                                ; %if.then
        call    abort
.Lfunc_end0:
        .size   test1, .Lfunc_end0-test1
                                        ; -- End function
        .globl  main                            ; -- Begin function main
        .type   main,@function
main:                                   ; @main
; %bb.0:                                ; %entry
        enter   r30,r0,0,8
        std     #1234,[sp]
        mov     r30,#0
        mov     r1,r30
        mov     r2,r30
        mov     r3,r30
        mov     r4,r30
        mov     r5,r30
        mov     r6,r30
        mov     r7,r30
        call    test1
        mov     r1,r30
        exit    r30,r0,0,8
```
Now, main stores the value 1234 on the stack, that looks OK. But what test1 is doing... it does not appear to load the value from the stack.  Rather, it adds 8 to the stack value, stores it on the stack, masks it with 0xfffffffc and then compares it against 1234. That does not look right.

bagel99, 2023-02-01:
Yes, that code looks wrong  I had tested against your "Shorter test case" and that code looked OK.  Varargs is one of
the least tested part of the compiler.

bagel99, 2023-02-09:
Try again with commit 5a26c794e6a7

tkoenig1, 2023-02-10:
Looking better, stack passing seems to be OK now (although I'd have to look how the stack is adjusted).

However, the constant that is passed to the varargs appears to be off:
```
$ cat x.c 
/* PR target/44942 */

#include <stdarg.h>

void
test3 (double a, double b, double c, double d, double e, double f,
       double g, long double h, ...)
{
  double i;
  va_list ap;

  va_start (ap, h);
  i = va_arg (ap, double);
  if (i != 1234.0)
    __builtin_abort ();
  va_end (ap);
}

int
main ()
{
  test3 (0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0L, 1234.0);
  return 0;
}
$ clang --target=my66000 -emit-llvm -c x.c
$ llc -march=my66000 x.bc
$ cat x.s
        .text
        .file   "x.c"
        .globl  test3                           ; -- Begin function test3
        .type   test3,@function
test3:                                  ; @test3
; %bb.0:                                ; %entry
        add     sp,sp,#-80
        std     r1,[sp,72]
        std     r2,[sp,64]
        std     r3,[sp,56]
        std     r4,[sp,48]
        std     r5,[sp,40]
        std     r6,[sp,32]
        std     r7,[sp,24]
        std     r8,[sp,16]
        add     r1,sp,#80
        std     r1,[sp]
        ldd     r1,[sp]
        add     r2,r1,#8
        std     r2,[sp]
        ldd     r1,[r1]
        std     r1,[sp,8]
        ldd     r1,[sp,8]
        fcmp    r1,r1,#0x449A4000
        bnne    r1,.LBB0_2
        br      .LBB0_1
.LBB0_1:                                ; %if.then
        call    abort
.LBB0_2:                                ; %if.end
        add     sp,sp,#80
        ret
.Lfunc_end0:
        .size   test3, .Lfunc_end0-test3
                                        ; -- End function
        .globl  main                            ; -- Begin function main
        .type   main,@function
main:                                   ; @main
; %bb.0:                                ; %entry
        enter   r0,r0,0,16
        stw     #0,[sp,12]
        std     #4653142004841054208,[sp]
        cvtsd   r1,#0
        mov     r2,r1
        mov     r3,r1
        mov     r4,r1
        mov     r5,r1
        mov     r6,r1
        mov     r7,r1
        mov     r8,r1
        call    test3
        mov     r1,#0
        exit    r0,r0,0,16
.Lfunc_end1:
        .size   main, .Lfunc_end1-main
                                        ; -- End function
        .ident  "clang version 15.0.0 (https://github.com/bagel99/llvm-my66000.git 5a26c794e6a7bbe1692c2c4c4399b95ef8124c16)"
        .section        ".note.GNU-stack","",@progbits
```

tkoenig1, 2023-02-10:
So, looks fixed. Great!

closed 2023-02-10 by tkoenig1