bagel99/llvm-my66000 #23: Cannot select smul_lohi

tkoenig1, 2023-01-22:
Here's another from the gcc testsuite, for __builtin_mul_overflow, apparently:
```
/* PR tree-optimization/64006 */

int v;

long __attribute__ ((noinline, noclone))
test (long *x, int y)
{
  int i;
  long s = 1;
  for (i = 0; i < y; i++)
    if (__builtin_mul_overflow (s, x[i], &s))
      v++;
  return s;
}

int
main ()
{
  long d[7] = { 975, 975, 975, 975, 975, 975, 975 };
  long r = test (d, 7);
  if (sizeof (long) * __CHAR_BIT__ == 64 && v != 1)
    __builtin_abort ();
  else if (sizeof (long) * __CHAR_BIT__ == 32 && v != 4)
    __builtin_abort ();
  return 0;
}
```
[pr64006.zip](https://github.com/bagel99/llvm-my66000/files/10474266/pr64006.zip)

bagel99, 2023-01-25:
Try again with commit c66d22534c5a05431beb6fbd7573c66aff89f12b

tkoenig1, 2023-01-25:
Works, closing.

closed 2023-01-25 by tkoenig1