bagel99/llvm-my66000 #24: Cannot select shl_parts

tkoenig1, 2023-01-22:
Another one from the gcc testsuite.
```
/* PR target/85582 */

#ifdef __SIZEOF_INT128__
typedef __int128 S;
typedef unsigned __int128 U;
#else
typedef long long S;
typedef unsigned long long U;
#endif

__attribute__((noipa)) U
f1 (U x, int y)
{
  return x << (y & -2);
}

__attribute__((noipa)) S
f2 (S x, int y)
{
  return x >> (y & -2);
}

__attribute__((noipa)) U
f3 (U x, int y)
{
  return x >> (y & -2);
}

int
main ()
{
  U a = (U) 1 << (sizeof (U) * __CHAR_BIT__ - 7);
  if (f1 (a, 5) != ((U) 1 << (sizeof (S) * __CHAR_BIT__ - 3)))
    __builtin_abort ();
  S b = (U) 0x101 << (sizeof (S) * __CHAR_BIT__ / 2 - 7);
  if (f1 (b, sizeof (S) * __CHAR_BIT__ / 2) != (U) 0x101 << (sizeof (S) * __CHAR_BIT__ - 7))
    __builtin_abort ();
  if (f1 (b, sizeof (S) * __CHAR_BIT__ / 2 + 2) != (U) 0x101 << (sizeof (S) * __CHAR_BIT__ - 5))
    __builtin_abort ();
  S c = (U) 1 << (sizeof (S) * __CHAR_BIT__ - 1);
  if ((U) f2 (c, 5) != ((U) 0x1f << (sizeof (S) * __CHAR_BIT__ - 5)))
    __builtin_abort ();
  if ((U) f2 (c, sizeof (S) * __CHAR_BIT__ / 2) != ((U) -1 << (sizeof (S) * __CHAR_BIT__ / 2 - 1)))
    __builtin_abort ();
  if ((U) f2 (c, sizeof (S) * __CHAR_BIT__ / 2 + 2) != ((U) -1 << (sizeof (S) * __CHAR_BIT__ / 2 - 3)))
    __builtin_abort ();
  U d = (U) 1 << (sizeof (S) * __CHAR_BIT__ - 1);
  if (f3 (c, 5) != ((U) 0x1 << (sizeof (S) * __CHAR_BIT__ - 5)))
    __builtin_abort ();
  if (f3 (c, sizeof (S) * __CHAR_BIT__ / 2) != ((U) 1 << (sizeof (S) * __CHAR_BIT__ / 2 - 1)))
    __builtin_abort ();
  if (f3 (c, sizeof (S) * __CHAR_BIT__ / 2 + 2) != ((U) 1 << (sizeof (S) * __CHAR_BIT__ / 2 - 3)))
    __builtin_abort ();
  return 0;
}
```
yields the error message
```
LLVM ERROR: Cannot select: 0x55f3c76ffcf8: i64,i64 = shl_parts 0x55f3c76ff9b8, 0x55f3c76ffa88, 0x55f3c7700310
  0x55f3c76ff9b8: i64,ch = CopyFromReg 0x55f3c76bb428, Register:i64 %0
    0x55f3c76ff950: i64 = Register %0
  0x55f3c76ffa88: i64,ch = CopyFromReg 0x55f3c76bb428, Register:i64 %1
    0x55f3c76ffa20: i64 = Register %1
  0x55f3c7700310: i64 = and 0x55f3c76ffc90, Constant:i64<4294967294>
    0x55f3c76ffc90: i64 = AssertSext 0x55f3c76ffb58, ValueType:ch:i32
      0x55f3c76ffb58: i64,ch = CopyFromReg 0x55f3c76bb428, Register:i64 %2
        0x55f3c76ffaf0: i64 = Register %2
    0x55f3c76ffe30: i64 = Constant<4294967294>
In function: f1
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.	Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 pr85582-3_opt.bc
1.	Running pass 'Function Pass Manager' on module 'pr85582-3_opt.bc'.
2.	Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@f1'
 #0 0x000055f3c38e365f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/asdf/bin/llc+0x2ee365f)
```
[pr85592-3.zip](https://github.com/bagel99/llvm-my66000/files/10474271/pr85592-3.zip)

```

tkoenig1, 2023-01-22:
Same with
```
#include <stdint.h>

typedef unsigned char uchar_t;

void
unpack (uchar_t size[], uint64_t packed[], uint64_t unpacked[], uint64_t count)
{
  uint64_t len, bit = 0, word = 1, container1, container2 = packed[0];

  for (unsigned int i = 0; i < count; i++)
    {
      container1 = container2;
      container2 = packed[word++];
      __uint128_t cont;
      cont = ((__uint128_t)container2 << 64) | container1;
      do
        {
          len = size[i];
          unpacked[i] = (cont >> bit) & ~(~0 << len);
          bit += len;
        }
      while (bit < 64);
      bit &= 63;
    }
}
```

tkoenig1, 2023-01-22:
One additional remark. Up to now, I've ben using Release compilers, now I switched to Debug (which, presumably, you are also using). This might explain some of the differences we have seen so far.

bagel99, 2023-01-22:
I do compile to Debug.  Compilation is much slower than Release and the final link step uses a lot of memory.  When I compile to Debug I limit the number of targets to 3 or 4.

bagel99, 2023-01-25:
Try again with commit c66d22534c5a05431beb6fbd7573c66aff89f12b

tkoenig1, 2023-01-25:
Works now. Thanks!

tkoenig1, 2023-01-25:
... and closing.

closed 2023-01-25 by tkoenig1