bagel99/llvm-my66000 #25: Assertion `!NodePtr->isKnownSentinel()' failed.

tkoenig1, 2023-01-23:
Here's an interesting one, reduced from gcc's 20040916-1.c:

```
marker, buff;
bar() {
  for (;;) {
    foo();
    switch (marker)
    case 4:
      buff++;
  }
}
```
Error message is:
```
lc: /home/asdf/llvm-my66000/llvm/include/llvm/ADT/ilist_iterator.h:138: llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::operator*() const [with OptionsT = llvm::ilist_detail::node_options<llvm::MachineBasicBlock, true, false, void>; bool IsReverse = false; bool IsConst = false; llvm::ilist_iterator<OptionsT, IsReverse, IsConst>::reference = llvm::MachineBasicBlock&]: Assertion `!NodePtr->isKnownSentinel()' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 20040916-1_opt.bc
1.      Running pass 'Function Pass Manager' on module '20040916-1_opt.bc'.
2.      Running pass 'My66000 VVM Loop Analysis' on function '@bar'
 #0 0x00005643ff88a71e llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /home/asdf/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:565:22
 #1 0x00005643ff88aae9 PrintStackTraceSignalHandler(void*) /home/asdf/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:632:1
 #2 0x00005643ff8883e8 llvm::sys::RunSignalHandlers() /home/asdf/llvm-my66000/llvm/lib/Support/Signals.cpp:97:20
 #3 0x00005643ff88a052 SignalHandler(int) /home/asdf/llvm-my66000/llvm/lib/Support/Unix/Signals.inc:407:1
 #4 0x00007f4644aee420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #5 0x00007f464458b00b raise /build/glibc-SzIz7B/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #6 0x00007f464456a859 abort /build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c:81:7
 #7 0x00007f464456a729 get_sysdep_segment_value /build/glibc-SzIz7B/glibc-2.31/intl/loadmsgcat.c:509:8
 #8 0x00007f464456a729 _nl_load_domain /build/glibc-SzIz7B/glibc-2.31/intl/loadmsgcat.c:970:34
 #9 0x00007f464457bfd6 (/lib/x86_64-linux-gnu/libc.so.6+0x33fd6)
#10 0x00005643fc638fd7 llvm::ilist_iterator<llvm::ilist_detail::node_options<llvm::MachineBasicBlock, true, false, void>, false, false>::operator*() const /home/asdf/llvm-my66000/llvm/include/llvm/ADT/ilist_iterator.h:139:33
#11 0x00005643fe7d876e llvm::MachineLoop::getBottomBlock() /home/asdf/llvm-my66000/llvm/lib/CodeGen/MachineLoopInfo.cpp:84:18
#12 0x00005643fdcd2c0e (anonymous namespace)::My66000VVMLoop::checkLoop(llvm::MachineLoop*) /home/asdf/llvm-my66000/llvm/lib/Target/My66000/My66000VVMPass1.cpp:86:28
#13 0x00005643fdcd4404 (anonymous namespace)::My66000VVMLoop::runOnMachineFunction(llvm::MachineFunction&) /home/asdf/llvm-my66000/llvm/lib/Target/My66000/My66000VVMPass1.cpp:315:24
#14 0x00005643fe7bb6a9 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) /home/asdf/llvm-my66000/llvm/lib/CodeGen/MachineFunctionPass.cpp:72:33
#15 0x00005643fee3715e llvm::FPPassManager::runOnFunction(llvm::Function&) /home/asdf/llvm-my66000/llvm/lib/IR/LegacyPassManager.cpp:1434:20
#16 0x00005643fee37427 llvm::FPPassManager::runOnModule(llvm::Module&) /home/asdf/llvm-my66000/llvm/lib/IR/LegacyPassManager.cpp:1480:13
#17 0x00005643fee37899 (anonymous namespace)::MPPassManager::runOnModule(llvm::Module&) /home/asdf/llvm-my66000/llvm/lib/IR/LegacyPassManager.cpp:1549:20
#18 0x00005643fee327ba llvm::legacy::PassManagerImpl::run(llvm::Module&) /home/asdf/llvm-my66000/llvm/lib/IR/LegacyPassManager.cpp:539:13
#19 0x00005643fee3816f llvm::legacy::PassManager::run(llvm::Module&) /home/asdf/llvm-my66000/llvm/lib/IR/LegacyPassManager.cpp:1677:1
#20 0x00005643fc398f3c compileModule(char**, llvm::LLVMContext&) /home/asdf/llvm-my66000/llvm/tools/llc/llc.cpp:722:66
#21 0x00005643fc396b81 main /home/asdf/llvm-my66000/llvm/tools/llc/llc.cpp:417:35
#22 0x00007f464456c083 __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:342:3
#23 0x00005643fc39593e _start (/home/asdf/bin/llc+0x1bab93e)
```
[knownsentinel.zip](https://github.com/bagel99/llvm-my66000/files/10477327/knownsentinel.zip)

bagel99, 2023-01-31:
Try again with commit 4c3bbcf161cfd1940c03640c831d1e2b24ba783d

tkoenig1, 2023-02-01:
Fixed, thanks!

closed 2023-02-01 by tkoenig1