bagel99/llvm-my66000 #26: Target does not support TLS or Common sections

tkoenig1, 2023-01-23:
Here's a short one:
```
$ cat Wc99-c11-compat-1.c
_Thread_local int i;
$ 66 Wc99-c11-compat-1.c 
LLVM ERROR: Target does not support TLS or Common sections
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 Wc99-c11-compat-1_opt.bc
```

bagel99, 2023-01-28:
Try again with commit 9f1575dce18acaa0c24c19b657259cd769c3962f

tkoenig1, 2023-01-28:
Works, great!

I'm not sure if you realized this, but I threw every C test in the gcc testsuite at your compiler, and filed all the my66000-specific bugs I found (surprisingly few, IMHO).  So, now that this burst is over, you can expect a lower frequency of bug reports again :-)

closed 2023-01-28 by tkoenig1