bagel99/llvm-my66000 #27: ICE in clang on promoted vararg

tkoenig1, 2023-01-23:
The code is doubious, a warning is expected.  No ICE with x86_64.

```
$ cat Wcxx-compat-11.c 
/* { dg-do compile } */
/* { dg-options "-Wc++-compat" } */

#include <stdarg.h>

enum E { A };

extern void f2 (int);
void
f1 (int n, ...)
{
  va_list ap;

  va_start (ap, n);
  f2 (va_arg (ap, int));
  f2 (va_arg (ap, _Bool));      /* { dg-warning "promoted" } */
  f2 (va_arg (ap, enum E));     /* { dg-warning "promoted" } */
}

/* Match extra informative notes.  */
/* { dg-message "note:" "expected" { target *-*-* } 0 } */
$ 66 Wcxx-compat-11.c 
Wcxx-compat-11.c:16:19: warning: second argument to 'va_arg' is of promotable type '_Bool'; this va_arg has undefined behavior because arguments will be promoted to 'int' [-Wvarargs]
  f2 (va_arg (ap, _Bool));      /* { dg-warning "promoted" } */
                  ^~~~~
/home/asdf/lib/clang/15.0.0/include/stdarg.h:19:50: note: expanded from macro 'va_arg'
#define va_arg(ap, type)    __builtin_va_arg(ap, type)
                                                 ^~~~
clang: /home/asdf/llvm-my66000/llvm/lib/IR/Instructions.cpp:1488: void llvm::StoreInst::AssertOK(): Assertion `cast<PointerType>(getOperand(1)->getType()) ->isOpaqueOrPointeeTypeMatches(getOperand(0)->getType()) && "Ptr must be a pointer to Val type!"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: clang -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize -emit-llvm -fno-unroll-loops Wcxx-compat-11.c
1.      <eof> parser at end of file
2.      Wcxx-compat-11.c:10:1: LLVM IR generation of declaration 'f1'
3.      Wcxx-compat-11.c:10:1: Generating code for declaration 'f1'
```

bagel99, 2023-01-23:
Can not duplicate.  Get the 2 warnings from clang and then things proceed normally.

tkoenig1, 2023-01-23:
> Can not duplicate. Get the 2 warnings from clang and then things proceed normally.

Here we go again...

[reproducer.zip](https://github.com/bagel99/llvm-my66000/files/10482616/reproducer.zip)

I have attached the two reproducer files that clang generates.  Maybe this will help (you will have
to adjust the paths, obviously :-)

bagel99, 2023-01-30:
Even with this new information, using your flags to clang, I cannot reproduce.
May post the *.bc so we can see if the IR differs somehow.

tkoenig1, 2023-01-30:
I cannot attach a *.bc file because the error is in clang, no *.bc is generated.

If you run the reproducer script file from the zip file, adjusting the path to clang, do you also not get the error?

But maybe it is just a clang error masked by some peculiarities in other implementations.

FWITW, here is a gdb session showing the occurence of the error on my system.

[gdbsession.txt](https://github.com/bagel99/llvm-my66000/files/10540474/gdbsession.txt)

However, I would not really classify this as a high-priority bug :-)

bagel99, 2023-01-30:
OK.  I think I know why we get different results.  You are running a Debug version of clang, I am running a Release version. In the Debug version, asserts are enabled.  They are not in the Release version.
See if you get this assert when running clang for other targets.

tkoenig1, 2023-01-30:
I just ran it on a machine where I have a Relase build enabled, and I do not see anything there.

I also checked a couple of other architectures, and the error is not there with the same Release compiler I use for my66000:

```
$ clang -S -O2 Wcxx-compat-11-409fb6.c 
Wcxx-compat-11.c:16:19: warning: second argument to 'va_arg' is of promotable type '_Bool'; this va_arg has undefined behavior because arguments will be promoted to 'int' [-Wvarargs]
  f2 (va_arg (ap, _Bool));      /* { dg-warning "promoted" } */
                  ^~~~~
/home/asdf/lib/clang/15.0.0/include/stdarg.h:19:50: note: expanded from macro 'va_arg'
#define va_arg(ap, type)    __builtin_va_arg(ap, type)
                                                 ^~~~
1 warning generated.
$ clang --target=powerpc64le-unknown-linux -S -O2 Wcxx-compat-11-409fb6.c 
Wcxx-compat-11.c:16:19: warning: second argument to 'va_arg' is of promotable type '_Bool'; this va_arg has undefined behavior because arguments will be promoted to 'int' [-Wvarargs]
  f2 (va_arg (ap, _Bool));      /* { dg-warning "promoted" } */
                  ^~~~~
/home/asdf/lib/clang/15.0.0/include/stdarg.h:19:50: note: expanded from macro 'va_arg'
#define va_arg(ap, type)    __builtin_va_arg(ap, type)
                                                 ^~~~
1 warning generated.
$ clang --target=my66000 -S -O2 Wcxx-compat-11-409fb6.c 
Wcxx-compat-11.c:16:19: warning: second argument to 'va_arg' is of promotable type '_Bool'; this va_arg has undefined behavior because arguments will be promoted to 'int' [-Wvarargs]
  f2 (va_arg (ap, _Bool));      /* { dg-warning "promoted" } */
                  ^~~~~
/home/asdf/lib/clang/15.0.0/include/stdarg.h:19:50: note: expanded from macro 'va_arg'
#define va_arg(ap, type)    __builtin_va_arg(ap, type)
                                                 ^~~~
clang: /home/asdf/llvm-my66000/llvm/lib/IR/Instructions.cpp:1488: void llvm::StoreInst::AssertOK(): Assertion `cast<PointerType>(getOperand(1)->getType()) ->isOpaqueOrPointeeTypeMatches(getOperand(0)->getType()) && "Ptr must be a pointer to Val type!"' failed.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace, preprocessed source, and associated run script.
Stack dump:
0.      Program arguments: clang --target=my66000 -S -O2 Wcxx-compat-11-409fb6.c
1.      <eof> parser at end of file
2.      Wcxx-compat-11.c:10:1: LLVM IR generation of declaration 'f1'
3.      Wcxx-compat-11.c:10:1: Generating code for declaration 'f1'
^C
```

tkoenig1, 2024-06-23:
Seems fixed also in Debug mode with the current compiler, I cannot reproduce it any longer.

closed 2024-06-23 by tkoenig1