bagel99/llvm-my66000 #28: Subtraction with 128-bit int: Cannot select subcarry

tkoenig1, 2023-01-23:
Reduced a bit from the gcc testsuite.
```
typedef __int128_t TItype;
TItype
__subvdi3 (TItype a, TItype b)
{
  TItype w;
  w = a - b;
  return w;
}
```
gets
```
LVM ERROR: Cannot select: t22: i64,i64 = subcarry t4, t8, t21:1
  t4: i64,ch = CopyFromReg t0, Register:i64 %1
    t3: i64 = Register %1
  t8: i64,ch = CopyFromReg t0, Register:i64 %3
    t7: i64 = Register %3
  t21: i64,i64 = usubo t2, t6
    t2: i64,ch = CopyFromReg t0, Register:i64 %0
      t1: i64 = Register %0
    t6: i64,ch = CopyFromReg t0, Register:i64 %2
      t5: i64 = Register %2
In function: __subvdi3
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 auto_opt.bc
1.      Running pass 'Function Pass Manager' on module 'auto_opt.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@__subvdi3
```

bagel99, 2023-01-23:
Test again with commit 6385ec062ccdd3350dcf89f4f2a3653449fb075c

tkoenig1, 2023-01-24:
Looks good:
```
        carry   r5,{O,IO}
        add     r1,r1,-r3
        add     r2,r2,-r4
        ret
```
Closing.

closed 2023-01-24 by tkoenig1