bagel99/llvm-my66000 #34: Multiplying integers by 3, 5 or 9

tkoenig1, 2023-02-08:
Everybody's favorite micro-optimization.  Probably saves some cycles, but, for My66000, no instructions.
```
unsigned int uitimes_3 (unsigned int a)
{
  return a * 3;
}

unsigned int uitimes_5 (unsigned int a)
{
  return a * 5;
}

unsigned int uitimes_9 (unsigned int a)
{
  return a * 9;
}
```
gets
```
        mul     r1,r1,#3
        srl     r1,r1,<32:0>
        ret
```
etc, but could use
```
        la      r1,[r1, r1<<1]
        srl     r1,r1,<32:0>
        ret
```
instead.

bagel99, 2023-02-09:
Do you know of any LLVM target that does this?

tkoenig1, 2023-02-09:
There is x86_64 [https://godbolt.org/z/c1YavEqhj](url), ARM v8, where they use add+shift [https://godbolt.org/z/j7TeoYexP](url).  RISC-V uses shift and add, two instructions [https://godbolt.org/z/67n4GnnP9](url).

bagel99, 2023-03-02:
Try again with commit af7ee91ac91db0ac489c686eb35daf15ceb1f32f

tkoenig1, 2023-03-03:
Implemented, also works for signed.

Great!

closed 2023-03-03 by tkoenig1