bagel99/llvm-my66000 #35: Wong code with many floating point constants

tkoenig1, 2023-02-09:
Here's a stress-test with many floating point constants, where the result appears to be wrong (many zeros in the constants for the FMA):
[fmac.zip](https://github.com/bagel99/llvm-my66000/files/10699887/fmac.zip)
```
$ cat tst.c
#define A(x) (1.234+(x)*(-0.321+(x)*(0.05413+(x)*(0.0841234+(x)*0.0001242+(x)*0.05421))))
#define B(x) (1.2414+(x)*(-0.351+(x)*(0.05113+(x)*(0.0841734+(x)*0.0001842+(x)*0.05421))))

double foo(double a, double b)
{
  return B(A(a+b));
}

$ clang --target=my66000 -emit-llvm -c tst.c
$ llc tst.bc
$ cat tst.s 
        .text
        .file   "tst.c"
        .globl  foo                             ; -- Begin function foo
        .type   foo,@function
foo:                                    ; @foo
; %bb.0:                                ; %entry
        add     sp,sp,#-16
        std     r1,[sp,8]
        std     r2,[sp]
        ldd     r1,[sp,8]
        ldd     r2,[sp]
        fadd    r1,r1,r2
        mov     r2,#0x0000000000000000
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r2,r1,#0x0000000000000000,r2
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r1,r1,r2,#0x3FF0000000000000
        mov     r2,#0x0000000000000000
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r2,r1,#0x0000000000000000,r2
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r2,r1,r2,#0x0000000000000000
        fmac    r1,r1,r2,#0x3FF0000000000000
        add     sp,sp,#16
        ret
.Lfunc_end0:
        .size   foo, .Lfunc_end0-foo
                                        ; -- End function
        .ident  "clang version 15.0.0 (https://github.com/bagel99/llvm-my66000.git 4c3bbcf161cfd1940c03640c831d1e2b24ba783d)"
        .section        ".note.GNU-stack","",@progbits
```

tkoenig1, 2023-02-09:
Reduced test case:
```
$ cat tst4.c 
double foo(double a)
{
  return (1.2414+((1.234+(a)*0.321)));
}
$ clang --target=my66000 -emit-llvm -c tst4.c
$ llc tst4.bc
$ cat tst4.
tst4.bc  tst4.c   tst4.s   
$ cat tst4.s 
        .text
        .file   "tst4.c"
        .globl  foo                             ; -- Begin function foo
        .type   foo,@function
foo:                                    ; @foo
; %bb.0:                                ; %entry
        add     sp,sp,#-8
        std     r1,[sp]
        ldd     r1,[sp]
        mov     r2,#0x0000000000000000
        fmac    r1,r1,r2,#0x3FF0000000000000
        fadd    r1,r1,#0x3FF0000000000000
        add     sp,sp,#8
        ret
.Lfunc_end0:
        .size   foo, .Lfunc_end0-foo
                                        ; -- End function
        .ident  "clang version 15.0.0 (https://github.com/bagel99/llvm-my66000.git 4c3bbcf161cfd1940c03640c831d1e2b24ba783d)"
        .section        ".note.GNU-stack","",@progbits
```

tkoenig1, 2023-02-09:
LLVM assembler is
```
define dso_local double @foo(double noundef %a) #0 {
entry:
  %a.addr = alloca double, align 8
  store double %a, double* %a.addr, align 8
  %0 = load double, double* %a.addr, align 8
  %1 = call double @llvm.fmuladd.f64(double %0, double 3.210000e-01, double 1.234000e+00)
  %add = fadd double 1.241400e+00, %1
  ret double %add
}
```
The two constants in the FMA may be the problem.

bagel99, 2023-02-09:
I discovered this recently, also.  Some change in the base code from release 12 to release 14 broke this.

bagel99, 2023-02-09:
Try again with commit 5a26c794e6a7

tkoenig1, 2023-02-11:
Yes, works now.  Thanks!

closed 2023-02-11 by tkoenig1