bagel99/llvm-my66000 #36: Jump tables with large offsets

tkoenig1, 2023-02-11:
Stress-testing the compiler revealed a potential bug, which is  unlikely to occur in practice often, so probably not a high priority.

The little Perl script
```
$ncall = 4000;
$ncase = 5;

for ($j=0; $j<$ncase;$j++) {
    for ($i=0; $i<$ncall; $i++) {
        print "void yyy_$j","_$i(int);\n";
    }
}
print <<"EOF";
void foo(int n, int m)
{
  switch (n)
    {
EOF

for ($j = 0; $j<$ncase; $j++)
{
    print "  case $j:\n";
    for ($i=0; $i<$ncall; $i++) 
    {
        print "    yyy_$j","_$i(m+$i+1);\n";
    }
    print "  break;\n";
}
print "  }\n}\n";
```
generates a C file with a switch statement with 4000 calls to
a function, resulting in 12000 instructions per case, for a total
of total of 60000 instructions. That is longer than what can
be expressed in a 16-bit jump table. (I used the calls because
it is an easy way to generate large sections of code that the
compiler cannot reduce). 

The generated assembly reads
```
foo:                                    ; @foo
; %bb.0:                                ; %entry
        enter   r30,r0,0,0
        mov     r30,r2
        cmp     r2,r1,#4
        bhi     r2,.LBB0_7
; %bb.1:                                ; %entry
        jtth    r1,#5
        .jt16   .LBB0_2,.LBB0_3,.LBB0_4,.LBB0_5,.LBB0_6,.LBB0_2
        .p2align 2
.LBB0_2:                                ; %sw.bb
        add     r1,r30,#1
        sra     r1,r1,<32:0>
        call    yyy_0_0
        add     r1,r30,#2
        sra     r1,r1,<32:0>
        call    yyy_0_1
        add     r1,r30,#3
        sra     r1,r1,<32:0>
        call    yyy_0_2
```
and so on, which uses a 16-bit jump table, if I read the assembly right.

I don't have the assembler running, this might actually something that
is fixed up there already, or something that could be done there - determining
sizes of instructions is what assemblers do.

gas contains some magic to do something similar (but far more ugly),
see https://ftp.gnu.org/old-gnu/Manuals/gas-2.9.1/html_chapter/as_7.html ,
section .word .

bagel99, 2023-02-13:
The assembler does not currently deal with this.  The LLVM way is to deal with it in LLVM, since (for some targets) they can output ELF objects.  AArch64 has a pass to shrink jump tables, but since ours are in the code section not read-only data, is isn't clear whether its applicable.

tkoenig1, 2024-06-23:
The assembler now fixes up the jump table instructions, so this is no longer an issue, I think.

closed 2024-06-23 by tkoenig1

tkoenig1, 2024-06-23:
... and the assembler also uses byte offsets if possible.

The method is to generate  8-bit offsets first, and if they don't fit, go to 16-bit offsets, and if these don't fit, 32-bit offsets.

There still is some work to do before I can push a new version of the assembler.