bagel99/llvm-my66000 #40: Bad machine code

tkoenig1, 2023-03-06:
Here' something that turned up when I re-ran compiling GSL.

Reduced source code:
```
typedef int size_t;
struct {
  size_t *data;
} * gsl_permutation_linear_cycles_p;
size_t gsl_permutation_linear_cycles_i, gsl_permutation_linear_cycles_k,
    gsl_permutation_linear_cycles_count;
size_t gsl_permutation_linear_cycles() {
  for (; gsl_permutation_linear_cycles_p; gsl_permutation_linear_cycles_i++) {
    gsl_permutation_linear_cycles_k =
        gsl_permutation_linear_cycles_p->data[gsl_permutation_linear_cycles_i];
    while (gsl_permutation_linear_cycles_k > gsl_permutation_linear_cycles_i)
      gsl_permutation_linear_cycles_k =
          gsl_permutation_linear_cycles_p
              ->data[gsl_permutation_linear_cycles_k];
    if (gsl_permutation_linear_cycles_k < gsl_permutation_linear_cycles_i)
      gsl_permutation_linear_cycles_count++;
  }
  return gsl_permutation_linear_cycles_count;
}
```
Error message:
```
# Machine code for function gsl_permutation_linear_cycles: NoPHIs, TracksLiveness, TiedOpsRewritten

bb.0.entry:
  successors: %bb.7(0x30000000), %bb.1(0x50000000); %bb.7(37.50%), %bb.1(62.50%)

  %0:gregs = LDDrr $r0, $r0, 0, @gsl_permutation_linear_cycles_p :: (dereferenceable load (s64) from @gsl_permutation_linear_cycles_p, !tbaa !2)
  BRC %bb.7, %0:gregs, 2
  BRU %bb.1

bb.1.for.body.lr.ph:
; predecessors: %bb.0
  successors: %bb.2(0x80000000); %bb.2(100.00%)

  %1:gregs = LDDri %0:gregs, 0 :: (load (s64) from %ir.data1, !tbaa !6)
  %15:gregs = LDUWrr $r0, $r0, 0, @gsl_permutation_linear_cycles_i :: (dereferenceable load (s32) from @gsl_permutation_linear_cycles_i, !tbaa !8)

bb.2.for.body:
; predecessors: %bb.1, %bb.6
  successors: %bb.3(0x80000000); %bb.3(100.00%)

  %16:gregs = COPY %15:gregs

bb.3.while.cond:
; predecessors: %bb.2, %bb.3
  successors: %bb.3(0x7c000000), %bb.4(0x04000000); %bb.3(96.88%), %bb.4(3.12%)

  %17:gregs = VEC 0
  %7:gregs = SRAri %16:gregs, 32, 0
  %16:gregs = LDSWrr %1:gregs, %7:gregs, 2, 0 :: (load (s32) from %ir.storemerge.in, !tbaa !8)
  STWrr %16:gregs, $r0, $r0, 0, @gsl_permutation_linear_cycles_k :: (store (s32) into @gsl_permutation_linear_cycles_k, !tbaa !8)
  %8:gregs = SRAri %15:gregs, 32, 0
  LOOP1ri 2, %16:gregs, %8:gregs, 0, %17:gregs, %bb.3

bb.4.while.end:
; predecessors: %bb.3
  successors: %bb.5(0x40000000), %bb.6(0x40000000); %bb.5(50.00%), %bb.6(50.00%)

  BRIB %bb.6, %9:gregs, 3
  BRU %bb.5

bb.5.if.then:
; predecessors: %bb.4
  successors: %bb.6(0x80000000); %bb.6(100.00%)

  %12:gregs = LDUWrr $r0, $r0, 0, @gsl_permutation_linear_cycles_count :: (dereferenceable load (s32) from @gsl_permutation_linear_cycles_count, !tbaa !8)
  %13:gregs = nsw ADDri %12:gregs, 1
  STWrr %13:gregs, $r0, $r0, 0, @gsl_permutation_linear_cycles_count :: (store (s32) into @gsl_permutation_linear_cycles_count, !tbaa !8)

bb.6.for.inc:
; predecessors: %bb.4, %bb.5
  successors: %bb.2(0x80000000); %bb.2(100.00%)

  %15:gregs = nsw ADDri %15:gregs, 1
  STWrr %15:gregs, $r0, $r0, 0, @gsl_permutation_linear_cycles_i :: (store (s32) into @gsl_permutation_linear_cycles_i, !tbaa !8)
  BRU %bb.2

bb.7.for.end:
; predecessors: %bb.0

  %14:gregs = LDSWrr $r0, $r0, 0, @gsl_permutation_linear_cycles_count :: (dereferenceable load (s32) from @gsl_permutation_linear_cycles_count, !tbaa !8)
  $r1 = COPY %14:gregs
  RET implicit killed $r1

# End machine code for function gsl_permutation_linear_cycles.

*** Bad machine code: Reading virtual register without a def ***
- function:    gsl_permutation_linear_cycles
- basic block: %bb.4 while.end (0x56240833eea8)
- instruction: BRIB %bb.6, %9:gregs, 3
- operand 1:   %9:gregs
LLVM ERROR: Found 1 machine code errors.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc -O2 --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 canonical_opt.bc
1.      Running pass 'Function Pass Manager' on module 'canonical_opt.bc'.
2.      Running pass 'Live Interval Analysis' on function '@gsl_permutation_linear_cycles'
 #0 0x000056240399753f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/asdf/bin/llc+0x33c153f)
 #1 0x0000562403994e2e SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f2285537420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #3 0x00007f2284fd400b raise /build/glibc-SzIz7B/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #4 0x00007f2284fb3859 abort /build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c:81:7
 #5 0x00005624038d2b27 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/asdf/bin/llc+0x32fcb27)
 #6 0x0000562402cd1daa llvm::MachineFunction::verify(llvm::Pass*, char const*, bool) const (/home/asdf/bin/llc+0x26fbdaa)
 #7 0x0000562402b54215 llvm::LiveRangeCalc::findReachingDefs(llvm::LiveRange&, llvm::MachineBasicBlock&, llvm::SlotIndex, unsigned int, llvm::ArrayRef<llvm::SlotIndex>) (/home/asdf/bin/llc+0x257e215)
 #8 0x0000562402b55cf2 llvm::LiveRangeCalc::extend(llvm::LiveRange&, llvm::SlotIndex, unsigned int, llvm::ArrayRef<llvm::SlotIndex>) (/home/asdf/bin/llc+0x257fcf2)
 #9 0x0000562402b56837 llvm::LiveIntervalCalc::extendToUses(llvm::LiveRange&, llvm::Register, llvm::LaneBitmask, llvm::LiveInterval*) (/home/asdf/bin/llc+0x2580837)
#10 0x0000562402b5732c llvm::LiveIntervalCalc::calculate(llvm::LiveInterval&, bool) (/home/asdf/bin/llc+0x258132c)
#11 0x0000562402b387bf llvm::LiveIntervals::computeVirtRegInterval(llvm::LiveInterval&) (/home/asdf/bin/llc+0x25627bf)
#12 0x0000562402b39eff llvm::LiveIntervals::computeVirtRegs() (/home/asdf/bin/llc+0x2563eff)
#13 0x0000562402b3d09f llvm::LiveIntervals::runOnMachineFunction(llvm::MachineFunction&) (/home/asdf/bin/llc+0x256709f)
#14 0x0000562402bee308 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x2618308)
#15 0x00005624030d6b95 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x2b00b95)
#16 0x00005624030d6dd9 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/asdf/bin/llc+0x2b00dd9)
#17 0x00005624030d7622 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/asdf/bin/llc+0x2b01622)
#18 0x00005624010a246b compileModule(char**, llvm::LLVMContext&) llc.cpp:0:0
#19 0x0000562400fc77e2 main (/home/asdf/bin/llc+0x9f17e2)
#20 0x00007f2284fb5083 __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:342:3
#21 0x0000562401099c6e _start (/home/asdf/bin/llc+0xac3c6e)
```
Reduction script:
```
STEM=canonical
clang -Og -fverbose-asm -c --target=my66000 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops -Werror $STEM.i || exit 1
llc --disable-lsr --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 $STEM.bc 2>&1 | grep "Bad machine"
```

bagel99, 2023-06-30:
I cannot reproduce with latest commit.  Please try again.

tkoenig1, 2023-07-03:
I still get the error:
```
asdf@flaemmli:~/GSL/PERM$ cat > p2.i
typedef int size_t;
struct
{
  size_t *data;
} * p;
size_t i, k, count;
size_t
foo ()
{
  for (; p; i++)
    {
      k = p->data[i];
      while (k > i)
        k = p->data[k];
      if (k < i)
        count++;
    }
  return count;
}
asdf@flaemmli:~/GSL/PERM$ clang -fverbose-asm -c --target=my66000 -O3 -fno-vectorize -fno-slp-vectorize  -emit-llvm -fno-unroll-loops -fomit-frame-pointer p2.i
asdf@flaemmli:~/GSL/PERM$ opt  -disable-loop-unrolling -O3  --march=my66000 --frame-pointer=none --enable-vvm p2.bc > p2_opt.bc
asdf@flaemmli:~/GSL/PERM$ llc -O2 -enable-remove-range-check --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 p2_opt.bc 

# Machine code for function foo: NoPHIs, TracksLiveness, TiedOpsRewritten

bb.0.entry:
  successors: %bb.7(0x30000000), %bb.1(0x50000000); %bb.7(37.50%), %bb.1(62.50%)

  %0:gregs = LDDrr $r0, $r0, 0, @p :: (dereferenceable load (s64) from @p, !tbaa !2)
  BRC %bb.7, %0:gregs, 2
  BRU %bb.1

bb.1.for.body.lr.ph:
; predecessors: %bb.0
  successors: %bb.2(0x80000000); %bb.2(100.00%)

  %1:gregs = LDDri %0:gregs, 0 :: (load (s64) from %ir.data1, !tbaa !6)
  %15:gregs = LDUWrr $r0, $r0, 0, @i :: (dereferenceable load (s32) from @i, !tbaa !8)

bb.2.for.body:
; predecessors: %bb.1, %bb.6
  successors: %bb.3(0x80000000); %bb.3(100.00%)

  %16:gregs = COPY %15:gregs

bb.3.while.cond:
; predecessors: %bb.2, %bb.3
  successors: %bb.3(0x7c000000), %bb.4(0x04000000); %bb.3(96.88%), %bb.4(3.12%)

  %17:gregs = VEC 0
  %7:gregs = SRAri %16:gregs, 32, 0
  %16:gregs = LDSWrr %1:gregs, %7:gregs, 2, 0 :: (load (s32) from %ir.storemerge.in, !tbaa !8)
  STWrr %16:gregs, $r0, $r0, 0, @k :: (store (s32) into @k, !tbaa !8)
  %8:gregs = SRAri %15:gregs, 32, 0
  LOOP1ri 2, %16:gregs, %8:gregs, 0, %17:gregs, %bb.3

bb.4.while.end:
; predecessors: %bb.3
  successors: %bb.5(0x40000000), %bb.6(0x40000000); %bb.5(50.00%), %bb.6(50.00%)

  BRIB %bb.6, %9:gregs, 3
  BRU %bb.5

bb.5.if.then:
; predecessors: %bb.4
  successors: %bb.6(0x80000000); %bb.6(100.00%)

  %12:gregs = LDUWrr $r0, $r0, 0, @count :: (dereferenceable load (s32) from @count, !tbaa !8)
  %13:gregs = nsw ADDri %12:gregs, 1
  STWrr %13:gregs, $r0, $r0, 0, @count :: (store (s32) into @count, !tbaa !8)

bb.6.for.inc:
; predecessors: %bb.4, %bb.5
  successors: %bb.2(0x80000000); %bb.2(100.00%)

  %15:gregs = nsw ADDri %15:gregs, 1
  STWrr %15:gregs, $r0, $r0, 0, @i :: (store (s32) into @i, !tbaa !8)
  BRU %bb.2

bb.7.for.end:
; predecessors: %bb.0

  %14:gregs = LDSWrr $r0, $r0, 0, @count :: (dereferenceable load (s32) from @count, !tbaa !8)
  $r1 = COPY %14:gregs
  RET implicit killed $r1

# End machine code for function foo.

*** Bad machine code: Reading virtual register without a def ***
- function:    foo
- basic block: %bb.4 while.end (0x5595ed4b8568)
- instruction: BRIB %bb.6, %9:gregs, 3
- operand 1:   %9:gregs
LLVM ERROR: Found 1 machine code errors.
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc -O2 -enable-remove-range-check --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 p2_opt.bc
1.      Running pass 'Function Pass Manager' on module 'p2_opt.bc'.
2.      Running pass 'Live Interval Analysis' on function '@foo'
 #0 0x00005595e7abde5f llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/asdf/bin/llc+0x33c1e5f)
 #1 0x00005595e7abb74e SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f27467aa420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #3 0x00007f274624700b raise /build/glibc-SzIz7B/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #4 0x00007f2746226859 abort /build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c:81:7
 #5 0x00005595e79f9447 llvm::report_fatal_error(llvm::Twine const&, bool) (/home/asdf/bin/llc+0x32fd447)
 #6 0x00005595e6df86ca llvm::MachineFunction::verify(llvm::Pass*, char const*, bool) const (/home/asdf/bin/llc+0x26fc6ca)
 #7 0x00005595e6c7ab35 llvm::LiveRangeCalc::findReachingDefs(llvm::LiveRange&, llvm::MachineBasicBlock&, llvm::SlotIndex, unsigned int, llvm::ArrayRef<llvm::SlotIndex>) (/home/asdf/bin/llc+0x257eb35)
 #8 0x00005595e6c7c612 llvm::LiveRangeCalc::extend(llvm::LiveRange&, llvm::SlotIndex, unsigned int, llvm::ArrayRef<llvm::SlotIndex>) (/home/asdf/bin/llc+0x2580612)
 #9 0x00005595e6c7d157 llvm::LiveIntervalCalc::extendToUses(llvm::LiveRange&, llvm::Register, llvm::LaneBitmask, llvm::LiveInterval*) (/home/asdf/bin/llc+0x2581157)
#10 0x00005595e6c7dc4c llvm::LiveIntervalCalc::calculate(llvm::LiveInterval&, bool) (/home/asdf/bin/llc+0x2581c4c)
#11 0x00005595e6c5f0df llvm::LiveIntervals::computeVirtRegInterval(llvm::LiveInterval&) (/home/asdf/bin/llc+0x25630df)
#12 0x00005595e6c6081f llvm::LiveIntervals::computeVirtRegs() (/home/asdf/bin/llc+0x256481f)
#13 0x00005595e6c639bf llvm::LiveIntervals::runOnMachineFunction(llvm::MachineFunction&) (/home/asdf/bin/llc+0x25679bf)
#14 0x00005595e6d14c28 llvm::MachineFunctionPass::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x2618c28)
#15 0x00005595e71fd4b5 llvm::FPPassManager::runOnFunction(llvm::Function&) (/home/asdf/bin/llc+0x2b014b5)
#16 0x00005595e71fd6f9 llvm::FPPassManager::runOnModule(llvm::Module&) (/home/asdf/bin/llc+0x2b016f9)
#17 0x00005595e71fdf42 llvm::legacy::PassManagerImpl::run(llvm::Module&) (/home/asdf/bin/llc+0x2b01f42)
#18 0x00005595e51c846b compileModule(char**, llvm::LLVMContext&) llc.cpp:0:0
#19 0x00005595e50ed7e2 main (/home/asdf/bin/llc+0x9f17e2)
#20 0x00007f2746228083 __libc_start_main /build/glibc-SzIz7B/glibc-2.31/csu/../csu/libc-start.c:342:3
#21 0x00005595e51bfc6e _start (/home/asdf/bin/llc+0xac3c6e)
Aborted (core dumped)
```

bagel99, 2023-07-04:
My mistake -- wrong options.  This error is caused by VVM optimization.

tkoenig1, 2024-06-22:
I can no longer reproduce that error, with or without --disable-slr, and the original source code, GSL, no longer shows it.

tkoenig1, 2024-06-22:
One click too many...

closed 2024-06-22 by tkoenig1