bagel99/llvm-my66000 #47: Implement fmod

tkoenig1, 2024-06-15:
```
double fmod(double,double);

double foo(double x)
{
        return fmod(x,4.0);
}
```
when compiled with -ffast-math (so instructions are generated instead of
library calls) currently fails with
```
66-fast fmod.c 
LLVM ERROR: Cannot select: 0x5aa87d12b580: f64 = frem nnan ninf nsz arcp contract afn reassoc 0x5aa87d12b4a0, 0x5aa87d12b7b0
  0x5aa87d12b4a0: f64,ch = CopyFromReg 0x5aa87d0ea350, Register:f64 %0
    0x5aa87d12b430: f64 = Register %0
  0x5aa87d12b7b0: f64 = My66000ISD::F64I5 Constant:i64<4>
    0x5aa87d12b740: i64 = Constant<4>
In function: foo
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.      Program arguments: llc -O2 --disable-lsr --enable-predication --enable-predication2 --enable-carry-generation --early-carry-coalesce --enable-vvm -march=my66000 fmod_opt.bc
1.      Running pass 'Function Pass Manager' on module 'fmod_opt.bc'.
2.      Running pass 'My66000 DAG->DAG Pattern Instruction Selection' on function '@foo'
```

bagel99, 2024-06-23:
Commit E1D8FA.. should fix this.

tkoenig1, 2024-06-23:
Yes, I now get

```
        carry   r2,{O}
        fdiv    r1,r1,#4
        mov     r1,r2
        ret
```

closed 2024-06-23 by tkoenig1