bagel99/llvm-my66000 #48: load/stores for type-punning unions

tkoenig1, 2024-07-20:
For the code
```
float inv_sqrt(float x)
{ union { float f; unsigned int u; } y = {x};
  y.u = 0x5F1FFFF9ul - (y.u >> 1);
  return 0.703952253f * y.f * (2.38924456f - x * y.f * y.f);
}
```
I get
```
inv_sqrt:                               ; @inv_sqrt
; %bb.0:                                ; %entry
        add     sp,sp,#-8
        stw     r1,[sp,4]
        lduw    r2,[sp,4]
        srl     r2,r2,#1,#0
        add     r2,#1595932665,-r2
        stw     r2,[sp]
        lduw    r2,[sp]
        fmulf   r3,r2,#0x3F343637
        fmulf   r1,r1,-r2
        fmacf   r1,r1,r2,#0x4018E962
        fmulf   r1,r3,r1
        add     sp,sp,#8
        ret
```
where the store/loads are apparently done for type punning.

There is probably something missing in the machine description that this conversion is, in fact, a no-op and needs no load/stores.

bagel99, 2024-07-25:
Possibly fixed by commit 3d9a2636b8d0.

tkoenig1, 2024-07-26:
> Possibly fixed by commit [3d9a263](https://github.com/bagel99/llvm-my66000/commit/3d9a2636b8d0949e98720eae845d2616cd8cdcbe).

Yes, the load/stores are gone, I now get
```
inv_sqrt:                               ; @inv_sqrt
; %bb.0:                                ; %entry
        and     r2,r1,#4294967294
        srl     r2,r2,#1,#0
        add     r2,#1595932665,-r2
        fmulf   r3,r2,#0x3F343637
        fmulf   r1,r1,-r2
        fmacf   r1,r1,r2,#0x4018E962
        fmulf   r1,r3,r1
        ret
```
As you remarked, the combination of and and srl looks weird (and it could
have been done with a single srl).

bagel99, 2024-07-30:
The and/srl appears to be a side effect of the type punning.  I can't reproduce it in "normal" code.

closed 2024-07-30 by bagel99